OttoDIY/OttoNinja #5: OTTO_calibration.ino won't compile / SerialUI

justinotherguy, 2017-06-11:
Hi,
first of all: thank you for this wonderful project! :)
I'd like to help improve - so here's a first issue report:
when I try to compile OTTO_calibration.ino I get a bunch of error messages (see below).
As far as I can tell, they seem to refer to SerialUI, which seems properly installed to me:

host:libraries justin$ ls -al SerialUI/
total 24
drwxr-xr-x   8 justin  staff   272  7 Jun 19:39 .
drwxr-xr-x  53 justin  staff  1802 11 Jun 09:07 ..
-rw-r--r--   1 justin  staff  3785 30 Nov  1979 README.md
drwxr-xr-x   4 justin  staff   136  7 Jun 19:39 examples
drwxr-xr-x   4 justin  staff   136  7 Jun 19:39 extras
-rw-r--r--   1 justin  staff  1256 30 Nov  1979 keywords.txt
-rw-r--r--   1 justin  staff   878 30 Nov  1979 library.properties
drwxr-xr-x  18 justin  staff   612  7 Jun 19:39 src
host:libraries justin$ pwd
/Users/justin/Documents/Arduino/libraries

Any idea, what I might be doing wrong?

My environment:
- Arduino 1.8.3 for OSX

Here's the output I get:

Arduino: 1.8.3 (Mac OS X), Board: "Arduino Nano, ATmega328"

/Users/justin/Documents/OttoDIY/OTTO_calibration/OTTO_calibration.ino: In function 'void setup()':
OTTO_calibration:102: error: no matching function for call to 'SUI::Menu::setName(const char [10])'
   mainMenu->setName(top_menu_title);
                                   ^
/Users/justin/Documents/OttoDIY/OTTO_calibration/OTTO_calibration.ino:102:35: note: candidate is:
In file included from /Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SerialUI.h:264:0,
                 from /Users/justin/Documents/Arduino/libraries/SerialUI/src/SerialUI.h:31,
                 from /Users/justin/Documents/OttoDIY/OTTO_calibration/OTTO_calibration.ino:24:
/Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SUIMenu.h:320:7: note: void SUI::Menu::setName(const __FlashStringHelper*)
  void setName(SOVA_FLASHSTRING_PTR namestr);
       ^
/Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SUIMenu.h:320:7: note:   no known conversion for argument 1 from 'const char [10]' to 'const __FlashStringHelper*'
OTTO_calibration:103: error: no matching function for call to 'SUI::Menu::subMenu(const char [3], const char [16])'
   SUI::Menu * laMenu = mainMenu->subMenu(la_key, la_help);
                                                         ^
/Users/justin/Documents/OttoDIY/OTTO_calibration/OTTO_calibration.ino:103:57: note: candidate is:
In file included from /Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SerialUI.h:264:0,
                 from /Users/justin/Documents/Arduino/libraries/SerialUI/src/SerialUI.h:31,
                 from /Users/justin/Documents/OttoDIY/OTTO_calibration/OTTO_calibration.ino:24:
/Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SUIMenu.h:308:9: note: SUI::Menu* SUI::Menu::subMenu(const __FlashStringHelper*, const __FlashStringHelper*)
  Menu * subMenu(SOVA_FLASHSTRING_PTR key_str, SOVA_FLASHSTRING_PTR help_str=NULL);
         ^
/Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SUIMenu.h:308:9: note:   no known conversion for argument 1 from 'const char [3]' to 'const __FlashStringHelper*'
OTTO_calibration:104: error: no matching function for call to 'SUI::Menu::addCommand(const char [2], void (&)(), const char [19])'
   laMenu->addCommand(la_plus_key, CB_leftAnklePlus, la_plus_help);
                                                                 ^
/Users/justin/Documents/OttoDIY/OTTO_calibration/OTTO_calibration.ino:104:65: note: candidate is:
In file included from /Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SerialUI.h:264:0,
                 from /Users/justin/Documents/Arduino/libraries/SerialUI/src/SerialUI.h:31,
                 from /Users/justin/Documents/OttoDIY/OTTO_calibration/OTTO_calibration.ino:24:
/Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SUIMenu.h:100:7: note: bool SUI::Menu::addCommand(const __FlashStringHelper*, SUI::MenuCommand_Callback, const __FlashStringHelper*)
  bool addCommand(SOVA_FLASHSTRING_PTR key_str, MenuCommand_Callback callback,SOVA_FLASHSTRING_PTR help_str=NULL);
       ^
/Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SUIMenu.h:100:7: note:   no known conversion for argument 1 from 'const char [2]' to 'const __FlashStringHelper*'
OTTO_calibration:105: error: no matching function for call to 'SUI::Menu::addCommand(const char [2], void (&)(), const char [19])'
   laMenu->addCommand(la_minus_key, CB_leftAnkleMinus, la_minus_help);
                                                                    ^
/Users/justin/Documents/OttoDIY/OTTO_calibration/OTTO_calibration.ino:105:68: note: candidate is:
In file included from /Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SerialUI.h:264:0,
                 from /Users/justin/Documents/Arduino/libraries/SerialUI/src/SerialUI.h:31,
                 from /Users/justin/Documents/OttoDIY/OTTO_calibration/OTTO_calibration.ino:24:
/Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SUIMenu.h:100:7: note: bool SUI::Menu::addCommand(const __FlashStringHelper*, SUI::MenuCommand_Callback, const __FlashStringHelper*)
[... 123 lines cut ...]
/Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SUIMenu.h:100:7: note:   no known conversion for argument 1 from 'const char [5]' to 'const __FlashStringHelper*'
OTTO_calibration:117: error: no matching function for call to 'SUI::Menu::addCommand(const char [5], void (&)(), const char [36])'
   mainMenu->addCommand(load_key, CB_load, load_help);
                                                    ^
/Users/justin/Documents/OttoDIY/OTTO_calibration/OTTO_calibration.ino:117:52: note: candidate is:
In file included from /Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SerialUI.h:264:0,
                 from /Users/justin/Documents/Arduino/libraries/SerialUI/src/SerialUI.h:31,
                 from /Users/justin/Documents/OttoDIY/OTTO_calibration/OTTO_calibration.ino:24:
/Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SUIMenu.h:100:7: note: bool SUI::Menu::addCommand(const __FlashStringHelper*, SUI::MenuCommand_Callback, const __FlashStringHelper*)
  bool addCommand(SOVA_FLASHSTRING_PTR key_str, MenuCommand_Callback callback,SOVA_FLASHSTRING_PTR help_str=NULL);
       ^
/Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SUIMenu.h:100:7: note:   no known conversion for argument 1 from 'const char [5]' to 'const __FlashStringHelper*'
OTTO_calibration:118: error: no matching function for call to 'SUI::Menu::addCommand(const char [6], void (&)(), const char [25])'
   mainMenu->addCommand(reset_key, CB_reset, reset_help);
                                                       ^
/Users/justin/Documents/OttoDIY/OTTO_calibration/OTTO_calibration.ino:118:55: note: candidate is:
In file included from /Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SerialUI.h:264:0,
                 from /Users/justin/Documents/Arduino/libraries/SerialUI/src/SerialUI.h:31,
                 from /Users/justin/Documents/OttoDIY/OTTO_calibration/OTTO_calibration.ino:24:
/Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SUIMenu.h:100:7: note: bool SUI::Menu::addCommand(const __FlashStringHelper*, SUI::MenuCommand_Callback, const __FlashStringHelper*)
  bool addCommand(SOVA_FLASHSTRING_PTR key_str, MenuCommand_Callback callback,SOVA_FLASHSTRING_PTR help_str=NULL);
       ^
/Users/justin/Documents/Arduino/libraries/SerialUI/src/includes/SUIMenu.h:100:7: note:   no known conversion for argument 1 from 'const char [6]' to 'const __FlashStringHelper*'
exit status 1
no matching function for call to 'SUI::Menu::setName(const char [10])'

hulkco, 2017-06-11:
Hi, try a old version of the IDE 1.6.X

justinotherguy, 2017-06-11:
Yep - using an old version of SerialUI helps! :)
I've downgraded SerialUI to 1.14.0 and now it works; versions 2.1.2 and 2.1.4 cause the above behaviour.
So - thx! Closing the issue as solved!

closed 2017-06-11 by justinotherguy