OttoDIY/OttoNinja #6: Zip file in Zip file - can we delete that?

justinotherguy, 2017-06-21:
Hi,

I see several ways to download the whole project in one file:
a) github -> download the file "OTTO_DIY_all.zip"
b) github -> using the green button "clone or download" -> "Download ZIP"

Is this intentional?
The disadvantages I see:
- using option b) (or by regularly cloning the project) I download every file twice, as the ZIP file is contained in the ZIP file respectively in the cloned git repo
- the content of the two directories is not (necessarily) identical in terms of versions

Regards, J.

cparrapa, 2017-06-21:
Hi justin
Yes thanks for the suggestion i also noticed that the thing is that is linked to our website for download from website. If you want to clone the project you can just delete the zip :)

justinotherguy, 2017-06-21:
well - that's true. It's a bit like "when you buy a beer, you also get a coffee; you can always spill that, if you don't like it", though ;-)
seriously: shouldn't the link on the website be updated to: https://github.com/OttoDIY/DIY/archive/master.zip ?

Regards, J.

cparrapa, 2017-06-21:
Is quite heavy because of the media part and the name is not the same as mentioned in the instruction manual, maybe will upload to another place like dropbox not sure for now hope doesn't disturb.

closed 2017-06-22 by cparrapa