OttoDIY/OttoNinja #15: _O defined as unsigned int in Oscillator.h

sushicodeur, 2017-11-08:
Hello,

I'm trying to make Otto work on my Esp8266.

It works quite well so far except for some very strange behaviors on oscillateServos function. For example if I run `Otto.updown(6, 500, BIG);` the right foot is totally up.

After some investigation it looks like for this function O is `int O[4] = {0, 0, h, -h};` so if you call updown with BIG you end up with these values `{0,0, 30, -30}`.

So far so good (well on arduino at last).

But on my ESP as _O is declared unsigned it takes 4,294,967,295 as a value... too much for my servo :)

So my question is : is it normal that _O is declared unsigned or is it a bug in Oscillator.h ?

cparrapa, 2017-11-09:
Hello
Awesome finally someone will make it!
hmmmm that value is to high maybe is a bug that wasn't noticeable in Arduino

Thanks!
Welcome to the Otto builder community!

sushicodeur, 2017-11-09:
I can confirm that changing the way _O is declared fixed the problem on my Esp Otto silly walk...

I now move on with the rest of the sketches and some more fun :)

I forked the repo here : https://github.com/sushicodeur/OttoOnEsp

cparrapa, 2017-11-10:
Great thanks! feel free to pull a request with a new folder or something i will just approve

sushicodeur, 2017-11-10:
I wait for my arduino nano to ship with ultrasonic sensor and shield, so I can test that my changes don't break on arduino.

cparrapa, 2017-11-10:
sure thanks for contributing! ;D

cparrapa, 2018-03-20:
Hey how is it going so is possible ESP with Otto?

closed 2018-05-10 by cparrapa