OttoDIY/OttoNinja #18: Missing LedMatrix.h

Pawka, 2018-03-11:
Compiling `Otto.h` file fails as `LedMatrix.h` is missing.

I haven't found this library anywhere on the repo and somewhere else. This might be an old library which was shipped with older Arduino IDE installations, but I was not able to find it there too.

Arduino IDE version: 1.8.5
Failing on: Otto.h:9 (`#include <LedMatrix.h>`)
OS: Ubuntu 16.04.4 LTS

## Workaround
As quick workaround would be disabling all calls to `LedMatrix` library.
After installing "Otto.zip" library, edit `Otto.h` and `Otto.cpp` files on  ~/Arduino/libraries/Otto/:

```diff
diff --git a/Otto.cpp b/Otto.cpp
index 2fa49f1..67bea9c 100644
--- a/Otto.cpp
+++ b/Otto.cpp
@@ -674,24 +674,24 @@ unsigned long int Otto::getAnimShape(int anim, int index){

 void Otto::putAnimationMouth(unsigned long int aniMouth, int index){

-      ledmatrix.writeFull(getAnimShape(aniMouth,index));
+      /* ledmatrix.writeFull(getAnimShape(aniMouth,index)); */
 }

 void Otto::putMouth(unsigned long int mouth, bool predefined){

   if (predefined){
-    ledmatrix.writeFull(getMouthShape(mouth));
+    /* ledmatrix.writeFull(getMouthShape(mouth)); */
   }
   else{
-    ledmatrix.writeFull(mouth);
+    /* ledmatrix.writeFull(mouth); */
   }
 }

 void Otto::clearMouth(){

-  ledmatrix.clearMatrix();
+  /* ledmatrix.clearMatrix(); */
 }

diff --git a/Otto.h b/Otto.h
index 7ed6116..2ce4018 100644
--- a/Otto.h
+++ b/Otto.h
@@ -6,7 +6,7 @@
 #include <EEPROM.h>

 #include <US.h>
-#include <LedMatrix.h>
+/* #include <LedMatrix.h> */
 #include <BatReader.h>

 #include "Otto_mouths.h"
@@ -95,7 +95,7 @@ class Otto

   private:

-    LedMatrix ledmatrix;
+    /* LedMatrix ledmatrix; */
     BatReader battery;
     Oscillator servo[4];
     US us;
```

cparrapa, 2018-03-20:
is ok we moved to another repository this is the basic entry Otto to make the code more simple

closed 2018-03-20 by cparrapa

Pawka, 2018-03-20:
@cparrapa In the first place I want to thank you for building and maintaining Otto!

You mentioned this repository is supposed to be an entry level for early beginners. Currently it can not be this entry level as people simply can not compile it. And it can act even in an opposite way - early beginners will be pushed off once they'll see they can not compile "entry level" repository.
When I was building the Otto, I searched how to solve this problem. What I've found was only complains that people can not make it work (e.g. see comments here): http://www.instructables.com/id/Otto-Build-You-Own-Robot-in-Two-Hours/

I see two ways to solve this:
- If this repository is outdated and not supposed to work, I'd suggest putting a note at the top of README.md and mention that original repository is somewhere else and this one is outdated.
- If current repository is supposed to work - make sure people can compile it. This can be done by merging #19 PR or refactoring code to get rid of `LetMatrix`.

Pawka, 2018-03-20:
> Hi pawka
> thanks fo the analysis is just that so many libraries, codes and people
> putting codes everywhere but not in github is making this out of control, i
> am not sure what is the problem, i personally deleted those LED matrix
> calls from the all the codes because we don't use anymore actually we have
> MaxMatrix
>
> In any case i let you merge the change in case you see more bugs will be
> really helpful to fix them
> Thanks!

Thank you for merging!

If LedMatrix is not used under this repo I will be glad to help you and
remove that unnecessary code. Just let me know.

cparrapa, 2018-03-20:
Thanks to you! we need more Otto builders like you, able to put some order to this repositories many bugs around and we are also testing new codes that need to be verified.

just pull requests for similar bugs and in that way we keep track of the changes