OttoDIY/OttoNinja #24: Error compiling for board arduino

freekvh, 2019-01-05:
# Issue Template

## Explain the problem:
I just got my Ottobot, it works well (was missing one screw though for the on/off button, it's also not listed on the part page of the manual by the way.
But I worked around that. I now have the following issue, some programs work, like Smooth criminal and all the "songs". But Otto avoid does not compile, I get the following error:

```

Arduino: 1.8.8 (Windows 10), Board: "Arduino Nano, ATmega328P"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\nlv24077\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10808 -build-path C:\Users\nlv24077\AppData\Local\Temp\arduino_build_577007 -warnings=none -build-cache C:\Users\nlv24077\AppData\Local\Temp\arduino_cache_726196 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\nlv24077\Documents\Arduino\Otto_avoid\Otto_avoid.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\nlv24077\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10808 -build-path C:\Users\nlv24077\AppData\Local\Temp\arduino_build_577007 -warnings=none -build-cache C:\Users\nlv24077\AppData\Local\Temp\arduino_cache_726196 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\nlv24077\Documents\Arduino\Otto_avoid\Otto_avoid.ino
Using board 'nano' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "C:\\Users\\nlv24077\\AppData\\Local\\Temp\\arduino_build_577007\\sketch\\Otto_avoid.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\libraries\\Servo\\src" "C:\\Users\\nlv24077\\AppData\\Local\\Temp\\arduino_build_577007\\sketch\\Otto_avoid.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\libraries\\Servo\\src" "-IC:\\Users\\nlv24077\\Documents\\Arduino\\libraries\\Oscillator" "C:\\Users\\nlv24077\\AppData\\Local\\Temp\\arduino_build_577007\\sketch\\Otto_avoid.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\libraries\\Servo\\src" "-IC:\\Users\\nlv24077\\Documents\\Arduino\\libraries\\Oscillator" "-IC:\\Users\\nlv24077\\Documents\\Arduino\\libraries\\US" "C:\\Users\\nlv24077\\AppData\\Local\\Temp\\arduino_build_577007\\sketch\\Otto_avoid.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\libraries\\Servo\\src" "-IC:\\Users\\nlv24077\\Documents\\Arduino\\libraries\\Oscillator" "-IC:\\Users\\nlv24077\\Documents\\Arduino\\libraries\\US" "-IC:\\Users\\nlv24077\\Documents\\Arduino\\libraries\\Otto" "C:\\Users\\nlv24077\\AppData\\Local\\Temp\\arduino_build_577007\\sketch\\Otto_avoid.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\libraries\\Servo\\src" "-IC:\\Users\\nlv24077\\Documents\\Arduino\\libraries\\Oscillator" "-IC:\\Users\\nlv24077\\Documents\\Arduino\\libraries\\US" "-IC:\\Users\\nlv24077\\Documents\\Arduino\\libraries\\Otto" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\EEPROM\\src" "C:\\Users\\nlv24077\\AppData\\Local\\Temp\\arduino_build_577007\\sketch\\Otto_avoid.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\libraries\\Servo\\src" "-IC:\\Users\\nlv24077\\Documents\\Arduino\\libraries\\Oscillator" "-IC:\\Users\\nlv24077\\Documents\\Arduino\\libraries\\US" "-IC:\\Users\\nlv24077\\Documents\\Arduino\\libraries\\Otto" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\EEPROM\\src" "-IC:\\Users\\nlv24077\\Documents\\Arduino\\libraries\\BatReader" "C:\\Users\\nlv24077\\AppData\\ [... cut ...]
Using library US in folder: C:\Users\nlv24077\Documents\Arduino\libraries\US (legacy)
Using library Otto in folder: C:\Users\nlv24077\Documents\Arduino\libraries\Otto (legacy)
Using library EEPROM at version 2.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM 
Using library BatReader in folder: C:\Users\nlv24077\Documents\Arduino\libraries\BatReader (legacy)
exit status 1
Error compiling for board Arduino Nano.
```

Can anyone help me?

By the way, the Otto is really cool! My wife will take it to her class of 8 year old soon...

letoangiakhanhg, 2019-03-03:
i have the same problem at my nano, what is your new code, pls help me

GalUnDrux, 2019-03-10:
I get the same error, even with the "updated" code.. which just replaces oscilator.h not Otto.h and Otto.cpp which contain the problem.. 
the problem seems to be with the LedMatrix entry within Otto.h.
No matter what i try i can not get any code to work with my laser cut v2 Otto Basic...
any help would be greatly apreciated!

GalUnDrux, 2019-03-10:
found a solution:
take the Otto folder from the App directory of the Otto-Plus Zip file from GitHub:
Otto-Plus\PLUS-master\APP\Libraries
copy all libraries to program files/Arduino/libraries (or your Arduino install directory)
it seems to have MaxMatrix rather than LEDMatrix inside the Otto.h... replacing the other Otto library directory seems to solve the compiling issue for Nano..

ProgrammerBruce, 2019-05-20:
Using Arduino IDE 1.8.9, without any changes, with the libraries copied from this repository to Documents/Arduino/libraries, I just confirmed that all of the .ino sketches compile successfully.  This includes:
Otto_allmoves
Otto_avoid_mhs
Otto_avoid_TRIMS
Otto_easycal
Otto_Footalign
Otto_happybirthday
Otto_jingle
Otto_MarioBros
Otto_Mblock_ScratchMode_Serial
Otto_push
Otto_run
Otto_singleladies
Otto_smooth_criminal
Otto_smooth_criminal_US

(Otto_ESP32 is a different issue.)

There may be an issue with differences between the .zip files and the other files in the libraries directory.  (The .zip files should probably be deleted from this repository, but that's a different issue.)

So, this issue #24 about a compiling problem should be closed.

cparrapa, 2019-05-20:
thanks @ProgrammerBruce for confirming that the codes work with no problems anymore

closed 2019-05-20 by cparrapa