OttoDIY/OttoNinja #25: Fit for symmetric foot unclear - detailed picture or graphic available?

justinotherguy, 2019-02-24:
Hello there, 

after printing the 3d files for OTTO it's unclear to me how the foot and the leg are supposed to fit together. Keeping to the diagrams in OttoDIY_InstructionsManual_V09.PDF on page 4 I end up with a leg that fits very loosely in the new leg (both are OttoDIY+_[foot|leg]_v08.stl).

Up until v5 the feet very asymmetrical with holes which very used to tightly mount the servo.
While I greatly appreciate the fact that the two feet are identical now, it seems I fail to understand how this is supposed to become stable.

Would you have any pictures that show details for an assembled foot + leg combination in v7 oder v8?

As I'm not suggesting an improvement but merely ask a question, let me know whether this (or thingiverse) is the right place.

Thanks guys :)

cparrapa, 2019-03-26:
Hi Justin
i will personally make a video to show this update the new feet design might not be the perfect fit specially for SG90 servos but allows to easily use any foot avoiding confusions, it seems that in some printers is ”quite loose“ so some already created an additional adapter to reinforce that will keep the compatibility with MG90 servos

thanks!

justinotherguy, 2019-03-28:
Hi @cparrapa ,
thx, great!
Maybe you can give the dimensions of the foot and the leg (the way they should be and do fit). So I can check whether it's my printer that  is so far off.
Thanks :)
Regards, J.

cparrapa, 2019-03-28:
Hi Justin
Have a look to the 3D model here https://gallery.autodesk.com/fusion360/projects/otto-diyplus
you can measure any part and distance, please check and let me know. Sometimes is the very tricky tolerances to make a perfect fit 
You are welcome

closed 2019-05-20 by cparrapa