OttoDIY/OttoNinja #34: Negative offset used for walking, but oscillator.h uses uint

Tobiassss, 2020-01-28:
My newly build OTTO couldn't walk, because it's right foot was not moving. It was just held in a fixed, angled position. Its right foot and all other servos worked (moved) correctly. By homing, all servos were correctly set into default position.

After some debugging, I could localize the problem: Oscillator::refresh calculated false huge numbers for the right foot servo, because the used offset (_O) was huge. It should be -4. The same with other negative values. Positives worked fine.
Digging deeper I found that  Oscillator.h only handles positive values:
`void SetO(unsigned int O) {_O=O;};`

By changing it to signed int, it works for me:
`void SetO(signed int O) {_O=O;};`

Did I miss something and my misbehaving OTTO has some other problems, or is this really an issue?

cparrapa, 2020-01-28:
Hi Tobias
Are you using ESP32 or ESP8266 board?

Tobiassss, 2020-01-29:
ESP32
TTGO T-Display (https://github.com/Xinyuan-LilyGO/TTGO-T-Display)

closed 2022-02-10 by cparrapa