OttoDIY/OttoNinja #36: Specified folder/zip files does not contain a valid library

Bmac947, 2020-04-29:
I'm getting this error when trying to use the Examples folder.

carl1961, 2020-04-29:
did you copy the Libraries folder to the ardunio libraries folder?    C:\Users\Carl\Documents\Arduino\libraries
[image]

Bmac947, 2020-04-29:
Thanks for the email! I have copied most of them after deleting and starting again,. The main issue in the Examples folder throwing the error!

[cid:9E77FDC3-7394-4833-9C93-A03A88ACE616@gateway]

[esp lib]<https://user-images.githubusercontent.com/3056821/80557213-c8cc7480-899b-11ea-9d31-4a308d1d9c42.png>

Ben McKenzie
Design Technology Coordinator
P:+61 3 5560 0888 | E: bmckenzie@emmanuel.vic.edu.au<mailto:bmckenzie@emmanuel.vic.edu.au>

[cid:image3c4a64.PNG@2aebdcf1.45b387db]
Emmanuel College

McAuley Campus 
37 Ardlie St 
Warrnambool, Victoria 3280      Rice Campus
42 Canterbury Rd
Warrnambool, Victoria 3280

www.emmanuel.vic.edu.au<http://www.emmanuel.vic.edu.au>
Warrnambool: Australia’s Most Liveable City<http://www.warrnambool.com/>

[cid:image2e51e8.JPG@41ad6a6c.4b8af60d]<https://www.emmanuel.vic.edu.au>
The information transmitted, including attachments, is intended only for the person(s) or entity to which it is addressed and may
contain confidential and/or privileged material. Any review, retransmission, dissemination or other use of, or taking of any action in
reliance upon this information by persons or entities other than the intended recipient is prohibited. If you received this in error
please contact the sender and destroy any copies of this information.

closed 2021-04-20 by cparrapa