OttoDIY/OttoNinja #38: Single Code Set: Merge OttoDIYESP & OttoDIYLib

sfranzyshen, 2020-07-31:
after looking at the differences between these two code sets ... it seems they could be merged into one. Is there interest in doing so? or should we continue developing for both?

Thanks #Ottobuilder

cparrapa, 2020-07-31:
It seems they are not compatible, we would have to first organize the naming, variables that use same command etc and compile- test them together, give a try and you will see, maybe you can help me to figure out how to make them compatible.

Only after we make sure it works with all Otto we can do a merge

sfranzyshen, 2020-07-31:
I will go ahead and take a shot at unifying them and see exactly where the conflicts are ...

cparrapa, 2020-07-31:
Goo idea thanks

sfranzyshen, 2020-09-22:
work has started for[ refactoring of the OttoDIY Robot API (Otto9.h) code to be non-blocking and universally cross-platform for all Arduino environments](https://github.com/sfranzyshen/OttoDIYArduino) ... so far the servo and motion functions have begun ...

sfranzyshen, 2020-09-27:
### we now have added the humanoid support 😄 

Refactoring of the OttoDIY Robot API (Otto9.h & Otto9Humanoid.h) code to be universally cross-platform (all-in-one) and 100% backwards compatible for all Arduino environments (Including but not limited to the Arduino Nano, UNO, Nano 33 IOT, Esp8266, and ESP32)

cparrapa, 2020-09-27:
Wow this is amazing how do you suggest we should merge this?

sfranzyshen, 2020-09-27:
> Wow this is amazing how do you suggest we should merge this in the repos?

ultimately, I could see replacing OttoDIYLIB & OttoDIYESP with OttoDIYArduino ...

cparrapa, 2020-09-28:
Hi @sfranzyshen 
Yes I think it can just merge all into the main OttoDIYLIB since many documentations have link connected to that and people always go there to download the libraries in that  repo and then we can get rid of the ESP one.

Later i would have to update for Otto Blockly

Then i think we should try to put it in the Arduino library manager so that is easy for people to install them and stay updated like any other arduino library https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

thanks

closed 2021-04-20 by cparrapa