OttoDIY/OttoNinja #43: Not an issue, just telling

jaiwiwjwjwisn, 2024-05-23:
Why not use esp32 as microcontroller? It has more ram, 4MB flash memory and built in wifi Bluetooth... That will be great.

jaiwiwjwjwisn, 2024-05-23:
Maybe we will also be able to ota update.

jaiwiwjwjwisn, 2024-05-23:
I found one but that's only shield and very outdated

cparrapa, 2024-05-28:
We never found a proper servo library for esp32 they all act very strange, any recommendation @jaiwiwjwjwisn ?

guavuavcycayfayqvqu, 2024-06-01:
> We never found a proper servo library for esp32 they all act very strange, any recommendation @jaiwiwjwjwisn ?

Please send repo /link I'm gonna check out.

guavuavcycayfayqvqu, 2024-06-01:
It's very painful to see a great project like otto can't use powerful microcontroller because of servo library.

guavuavcycayfayqvqu, 2024-06-01:
> We never found a proper servo library for esp32 they all act very strange, any recommendation @jaiwiwjwjwisn ?

Bytheway  rpi pico is also powerful and has 2MB flash (16MB in clones) and almost same size as arduino nano.  Best part is it has a good servo Library, please checkout.  And  unleash true potential of otto 🥰.

guavuavcycayfayqvqu, 2024-06-01:
Any thoughts @cparrapa ? https://github.com/guavuavcycayfayqvqu/OttoNinja/tree/devel/servo

sfranzyshen, 2024-06-02:
Hi @guavuavcycayfayqvqu or jaiwiwjwjwisn

We would like to help you. Please see the information in the OttoDIYLib comment https://github.com/OttoDIY/OttoDIYLib/issues/70

cparrapa, 2024-06-06:
It is a pity indeed @guavuavcycayfayqvqu but once you test them in the hardware you will see what I meant, @sfranzyshen  is doing a good work to have the compatibility and performance qual for nano and ESP boards.

There are many:
https://madhephaestus.github.io/ESP32Servo/annotated.html 
https://github.com/jkb-git/ESP32Servo
https://github.com/RoboticsBrno/ServoESP32 we were also testing and it was giving some errors and abrupt movements

They need to be properly merged with the Otto DIY Lib in a ways that it keeps backwards compatibility.

I also forgot to mention hardware was limitation until the new https://hprobots.com/ kit came out that one is ESP32 based and has the perfect size for Otto.